awslabs/amazon-kinesis-video-streams-producer-sdk-java #93: java-install-script not found

kennycoder, 2020-04-28:
Running Dockerfile build issues an error:
chmod: cannot access './java-install-script': No such file or directory

This file cannot be found in any previous releases nor in related repo:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp

disa6302, 2020-04-28:
@kennycoder , 

Last week, we moved to a cmake build system in producer CPP SDK and have not updated the Dockerfile for java yet. We will work on updating the docker file. Thank you for your patience in the meantime!

disa6302, 2020-04-28:
@kennycoder , 

I have pushed a change to solve this issue. You can grab the latest master and try it out. 
Resolving this for now. Feel free to open a new issue if you face any problems!

PR #94 (2020-04-28): Change dockerscript to adapt to cpp sdk changes
*Issue #, if available: #93 *

*Description of changes:*

Modified dockerfile to be in sync with latest changes in producer cpp sdk

Resolves #93 

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
closed 2020-04-28 by disa6302