awslabs/amazon-kinesis-video-streams-producer-sdk-java #95: embed audio to video streaming into aws kinesis through webcam

awslearne, 2020-04-29:
How to embed audio to a video streaming into aws kinesis through webcam.

MushMal, 2020-04-29:
It’s not clear what the question is but my guess is that you are asking about how to stream audio video from the attached camera using Java Producer Sdk? There is no common Java API to retrieve frames from the camera and encore them. Some customers use 3rd party libraries like JavaCV you could use and then use some 3rd party encoder to encode the frames and then push to Producer SDK. Same with audio, you might need to use some 3rd party frameworks. 

An alternative might be to use Java GStreamer to do the frame capture and the encoding.

awslearne, 2020-04-30:
I have completed capturing video and sending to aws kinesis by using webcam capture api
by using below link:
https://github.com/backdoorcodr/amazon-kinesis-video-streams-producer-sdk-java

But i also want to embed audio to streaming video created from above link,So im using below reference link to capture audio
https://github.com/ryancheu/webcam-capture/commit/25d63d6c822a5aa3483e22c0ebb16f889276b666

But when i start streaming audio and video i am not able to do it .when i look in my aws console im able to see video but no audio.

MushMal, 2020-05-06:
@awslearne sorry for the delay. To give a little background. The console playback is a "live" playback - meaning it operates on the fragments. It prepares MP4 fragments which are then played back using your browsers MSE. The playback only happens with multi-track stream. There are few things that might be wrong:

* Audio is not actually being embedded as a separate track.
* Audio that's embedded is not AAC or PCM that's supported in the console playback
* Your browser does not support the audio codec
* The content type is not reflecting the audio.

I believe it might be the latter. Please ensure your content type on the stream is set to something like "video/h264;audio/aac" or the corresponding lines for the codecs you are using.

You can set it directly in the AWS console.

awslearne, 2020-05-07:
is there any example program to capture live audio and video in multi-track stream using webcam api and sending it to aws kinesis?

MushMal, 2020-05-07:
I don't believe we have a sample for Java, there are some GStreamer based ones in C Producer

zhiyua-git, 2020-05-07:
@awslearne , if you can gather the audio as AAC then just for the multi-track part you can refer to [AudioVideoFrameSource](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/java/mediasource/file/AudioVideoFrameSource.java), [AudioVideoFileMediaSourceConfiguration](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/java/mediasource/file/AudioVideoFileMediaSourceConfiguration.java) and [AudioVideoFileMediaSource](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/java/mediasource/file/AudioVideoFileMediaSource.java). You don't need to deal with the thread sleep between putting audio/video frames and the file related operations.

awslearne, 2020-05-12:
@zhiyua-git 
I have few queries,even though im following your suggestion i have stucked in few place ie,
1.How to encode audio samples in pcm signed obtained from java sound api to AAC format in java.
2.Not able to send/combine audio and video as tracks

looking forward for your help.

MushMal, 2020-05-20:
@awslearne you are asking two different questions

1) You need to find an encoder that will encode AAC - we don't have any particular recommendation but perhaps you could look it up
2) Please cut a separate issue with detailed problem why you are unable to specify multi-track stream and what the errors are.

I am resolving this for now - please reopen or cut a new issue with particular question that needs addressing

closed 2020-05-20 by MushMal