awslabs/amazon-kinesis-video-streams-producer-sdk-java #97: video play in aws console only for 4 to 5 secs

awslearne, 2020-05-13:
Im following code as in 
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/10

Im streaming video from webcam to aws kinesis video.
im using below configuration :
	final CameraMediaSourceConfiguration configuration =
    			new CameraMediaSourceConfiguration.Builder()
    			.withFrameRate(FPS_19)
    			.withRetentionPeriodInHours(1)
    			.withCameraId("video0")
    			.withIsEncoderHardwareAccelerated(false)
    			.withEncodingMimeType("video/avc")
    			.withNalAdaptationFlags(StreamInfo.NalAdaptationFlags.NAL_ADAPTATION_ANNEXB_NALS)
    			.withIsAbsoluteTimecode(false)
    			.withEncodingBitRate(200000)//9000000
    			.withHorizontalResolution(640)
    			.withVerticalResolution(480)
    			.withCodecPrivateData(cpd)
    			.build();

but when i view video in aws console playback it show video only for few secs ie around 5 sec with lot of buffering.

How to stream and playback for more than 2 hrs??
Am i doing wrong in bit rate and frame rate??

MushMal, 2020-05-13:
@awslearne it seems that you have an issue on the producer side - could you please include debug logs?

awslearne, 2020-05-14:
please find log attached
[log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/4626577/log.txt)

MushMal, 2020-05-14:
The attached log is just the tail and it shows that your producer is constantly dropping frames

2020-05-14 06:28:30 viewItemRemoved(): Reporting a dropped frame/fragment.

It's observed bandwidth is over 30mbps.

Can you perhaps check the logs as you start - perhaps there is an error that is not included in the tail logs you've attached that can show the root cause

MushMal, 2020-05-20:
Resolving due to staleness. Please feel free to reopen or cut a new one to get further assistance

closed 2020-05-20 by MushMal