awslabs/amazon-kinesis-video-streams-producer-sdk-java #99: create new stream using AmazonKinesisVideoClient

awslearne, 2020-05-30:
Not able to create new stream using AmazonKinesisVideoClient class from aws-java-sdk-bom (1.11.602)
Getting error as:
[AV.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/4705257/AV.log)

According to attached log,im not able to find root cause for getting error in service name : "AWSAcuityReader" .
Im using correct credentials and connected to proper Vpc.

Any solution for this issue?

lherman-cs, 2020-06-01:
That definitely looks like a bug. @disa6302 maybe can comment on this?

MushMal, 2020-06-01:
@awslearne could you please include a little more information on your scenario? What repository and which asset/sample are you using and whether you made any changes locally?

MushMal, 2020-06-26:
Resolving as stale

closed 2020-06-26 by MushMal