awslabs/amazon-kinesis-video-streams-producer-sdk-java #103: AmazonKinesisVideoException: Service call failed

Newo6174, 2020-06-12:
ERROR / KinesisVideo: 2020-06-12T12:54Z T14: EXCEPTION: AmazonKinesisVideoException: Service call failed.: The security token included in the request is invalid. (Service: AmazonKinesisVideo; Status Code: 403; Error Code: UnrecognizedClientException; Request ID: a231f81c-9b6f-4239-a831-fb02fce5ada4)
ERROR / KinesisVideo: Kinesis Video service client returned an error com.amazonaws.services.kinesisvideo.model.AmazonKinesisVideoException: The security token included in the request is invalid. (Service: AmazonKinesisVideo; Status Code: 403; Error Code: UnrecognizedClientException; Request ID: a231f81c-9b6f-4239-a831-fb02fce5ada4). Reporting to Kinesis Video PIC.

I copy the access key from the csv so it should not be typo mistakes

MushMal, 2020-06-12:
Are you sure you are using access key and secret access key for the right region? The default region is us-west-2

Newo6174, 2020-06-12:
> Are you sure you are using access key and secret access key for the right region? The default region is us-west-2

I think so.
It shall be based on the KVS's region right?
I tried both US_WEST_2 and AP_NORTHEAST_1 but none of them work.

MushMal, 2020-06-12:
If you are using access key, secret access key combination it should work with any region. However, in your case for some reason those are invalid. Do you get temporary credentials? If so, there should also be a session token as well that you need to specify

Newo6174, 2020-06-12:
> If you are using access key, secret access key combination it should work with any region. However, in your case for some reason those are invalid. Do you get temporary credentials? If so, there should also be a session token as well that you need to specify

Hi, turns out it's typos 
I have included the <> in my access key.
But now I have encounter another problem:

DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
INFO / KinesisVideo: Initializing Kinesis Video client
DEBUG / KinesisVideo: calling describe stream: {StreamName: test-connect-jp-contact-0ec397d3-8851-4c71-ba9e-d4d9918d9d72,}
#
# There is insufficient memory for the Java Runtime Environment to continue.
# Native memory allocation (malloc) failed to allocate 133456 bytes for Chunk::new
# An error report file with more information is saved as:
# C:\Users\owenc\amazon-kinesis-video-streams-producer-sdk-java\hs_err_pid16712.log
[thread 2604 also had an error]
#
# Compiler replay data is saved as:
# C:\Users\owenc\amazon-kinesis-video-streams-producer-sdk-java\replay_pid16712.log

MushMal, 2020-06-12:
Perfect.

Not sure what this error is and why you are out of memory. Let's start with trying to get more information what you are trying to run on what platform and what hardware. I am not sure what Chunk::new is - doesn't seem related to KVS

closed 2020-06-12 by Newo6174