awslabs/amazon-kinesis-video-streams-producer-sdk-java #105: generate HLS url from Aws Kinesis video stream using java sdk

amiya-elear, 2020-07-06:
How can I generate an HLS link for live stream using Java SDK?
step 1 : to get data endpont I have written code to generate data endpoint in java kinesis 

final AmazonKinesisVideo frontendClient = AmazonKinesisVideoAsyncClient.builder()
                    .withCredentials(AuthHelper.getSystemPropertiesCredentialsProvider())
                    .withRegion(DEFAULT_REGION)
                    .build();
            System.out.println("data end point************&&&");
            /* this is the endpoint returned by GetDataEndpoint API */
            final String dataEndpoint = frontendClient.getDataEndpoint(
                    new GetDataEndpointRequest()
                            .withStreamName(STREAM_NAME)
                            .withAPIName("GET_HLS_STREAMING_SESSION_URL")).getDataEndpoint();
            System.out.println("data end point************"+ dataEndpoint);

step2: use GetHLSStreamingSessionURLRequest  to get HLS url for live stream

lherman-cs, 2020-07-06:
@amiya-elear you can modify your AmazonKinesisVideoArchivedMediaClient to be:

```java
final AmazonKinesisVideoArchivedMedia gethlsurl = AmazonKinesisVideoArchivedMediaClient.builder()
.withCredentials(AuthHelper.getSystemPropertiesCredentialsProvider())
.withEndpointConfiguration(new AwsClientBuilder.EndpointConfiguration(dataEndpoint, DEFAULT_REGION)) // <-- Notice that there's no  .withRegion(DEFAULT_REGION)
.build();
```

amiya-elear, 2020-07-07:
Thank you so much Lukas Herman, after passing retrieved dataEndpoint I am able to get HLS URL,

I have one small question now I am reading audio and video data from a single file frame by frame as 

private static final String FRAME_DIR = "src/main/resources/data/audio-video-frames";
private static MediaSource createFileMediaSource() {
        final AudioVideoFileMediaSourceConfiguration configuration =
                new AudioVideoFileMediaSourceConfiguration.AudioVideoBuilder()
                        .withDir(FRAME_DIR)
                        .withRetentionPeriodInHours(RETENTION_ONE_HOUR)
                        .withAbsoluteTimecode(ABSOLUTE_TIMECODES)
                        .withTrackInfoList(DemoTrackInfos.createTrackInfoList())
                        .build();
        final AudioVideoFileMediaSource mediaSource = new AudioVideoFileMediaSource(STREAM_NAME);
        mediaSource.configure(configuration);
        return mediaSource;

I want to read audio and video data from 2 separate data point means buffer where my camera witing audio and video, I want to read those two buffer pointer and write audio and video data (encoded audio and video file with timestamp)  into Kinesis video stream,  can you help me how can I do that using kinesis java producer sdk if you have any document with sample example you can share I will go through and implement the same.

lherman-cs, 2020-07-07:
@amiya-elear since the original question has been answered, I'll mark this issue as resolve. To help a larger community, I'd like to separate these 2 issues so that people can find the answer easier, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/106.

closed 2020-07-07 by lherman-cs