awslabs/amazon-kinesis-video-streams-producer-sdk-java #106: How to upload video and audio from 2 different sources

lherman-cs, 2020-07-07:
This is a continuation of https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/105

```
I have one small question now I am reading audio and video data from a single file frame by frame as

private static final String FRAME_DIR = "src/main/resources/data/audio-video-frames";
private static MediaSource createFileMediaSource() {
final AudioVideoFileMediaSourceConfiguration configuration =
new AudioVideoFileMediaSourceConfiguration.AudioVideoBuilder()
.withDir(FRAME_DIR)
.withRetentionPeriodInHours(RETENTION_ONE_HOUR)
.withAbsoluteTimecode(ABSOLUTE_TIMECODES)
.withTrackInfoList(DemoTrackInfos.createTrackInfoList())
.build();
final AudioVideoFileMediaSource mediaSource = new AudioVideoFileMediaSource(STREAM_NAME);
mediaSource.configure(configuration);
return mediaSource;

I want to read audio and video data from 2 separate data point means buffer where my camera witing audio and video, I want to read those two buffer pointer and write audio and video data (encoded audio and video file with timestamp) into Kinesis video stream, can you help me how can I do that using kinesis java producer sdk if you have any document with sample example you can share I will go through and implement the same.
```

lherman-cs, 2020-07-07:
Unfortunately, we don't have an example in Java, the closest one we have is https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_multistream_sample.cpp. This example is written in c++ and GStreamer for the media pipeline.

I would say the best way to achieve what you want is to implement a custom `AudioVideoFrameSource`, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/java/com/amazonaws/kinesisvideo/java/mediasource/file/AudioVideoFrameSource.java. Then, you would need to stitch the video data and audio data per frame. You would need to also be careful with the audio and video synchronization, make sure that the timestamps are aligned.

amiya-elear, 2020-07-08:
Thank you so much Lukas Herman, appreciate your help.

amiya-elear, 2020-07-22:
Hi Lukas Herman, Can we fork this repo "awslabs /amazon-kinesis-video-streams-producer-sdk-cpp"  write our own Java wrapper on top of that and use for stream data from a camera to kinesis.

lherman-cs, 2020-07-22:
@amiya-elear sure, as long as you follow Apache 2.0 license rules, https://www.apache.org/foundation/license-faq.html#Distribute-changes. 

But, doing this also means that you need to write your own bridge, very likely with JNI.

MushMal, 2020-07-22:
@amiya-elear you really don't need to fork the C++ repository. All you need is your own package that has it's own JNI transitions similar to the JNI native counterpart in the C++ repository that you can expose your own interfaces. You can take an example of how it's done here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/src/JNI

amiya-elear, 2020-07-23:
Thanks, Mushegh, what is the best way to write camera stream data into kinesis using java producer? as there is no direct way to do it. My camera provide me audio and video data in 2 channel (audio and video track) I just want to read those data create separate audio and separate video frame then write to in kinesis using java producer SDK.

MushMal, 2020-07-23:
If you really have the encoded audio and video frames given to you by your media pipeline, you don't really need to modify the JNI or come up with any special handling. The sample the @lherman-cs referenced has a commented line for audio/video multi-track stream. You might want to modify the code to use your own https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/fefd48b5229aa3237af3d2db31fad8f73880f15c/src/main/java/com/amazonaws/kinesisvideo/internal/mediasource/multitrack/MultiTrackFrameSource.java#L21

amiya-elear, 2020-07-23:
Thanks, MushMal for your response. 
Can you please give some insight on how to use this  MultiTrackFrameSource to pass frame buffer ? In my case, I have my own media API to get audio and video data from a camera in 2 channel and I have 2 call back function (audio_callback, video_callback) in call back function I will write video and audio frame into kinesis. can I achieve it using MultiTrackFrameSource? if not can you guide what is the best way to achieve it?

MushMal, 2020-07-23:
Well, if you have callbacks that means the "liveness" is on the media pipeline side - it will call your callbacks on a thread to act upon. The MultiTrackFrameSource is a dummy sample generator that generates frames and waits for a while to simulate the pipeline.

You might want to create your own class similar to multi track frame source and use it in the rest of the application that's modeled after the sample @lherman-cs mentioned.

In your case, whenever you get your callback, you call onDataAvailable and submit the furnished Frame object with the bits and flags of your media.

amiya-elear, 2020-07-24:
Do I need to create a MultiTrackMediaSource and register this MediaSource to kinesis video client? if yes how can I pass this frame object to MultiTrackMediaSourceConfiguration?

amiya-elear, 2020-07-24:
How can I submit received data in the callback to create a frame and put data into kinesis?

MushMal, 2020-07-24:
You can create your own media source and overwrite the onStreamDataAvailable. 

First store the interface pointer like so: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/fefd48b5229aa3237af3d2db31fad8f73880f15c/src/main/java/com/amazonaws/kinesisvideo/internal/mediasource/multitrack/MultiTrackFrameSource.java#L47

Then, when you have the frame to publish, call it like so:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/fefd48b5229aa3237af3d2db31fad8f73880f15c/src/main/java/com/amazonaws/kinesisvideo/internal/mediasource/multitrack/MultiTrackFrameSource.java#L82

amiya-elear, 2020-07-27:
Thank you so much Mushegh,
 Now I am able to send kinesis video frame (h264) in callback function by extending  MultiTrackFrameSource. I have small doubt is it mandatory to provide track info while creating MediaSource? 
In my case for video(format.h264) I have have used the same codecId and data to create video trackInfo provided in aws example. https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/fefd48b5229aa3237af3d2db31fad8f73880f15c/src/main/demo/com/amazonaws/kinesisvideo/demoapp/contants/DemoTrackInfos.java#L12
But from my camera I am getting audio (format PCMA audio.g711a)  not ACC to generate trackinfo for audio (format PCMA audio.g711a) what codecId and data I have to use?

MushMal, 2020-07-28:
There are some mappings in StreamInfo.java file. If you need G711 then you could simply go ahead with that track info but if you need AAC then you will need to re-encode the audio

amiya-elear, 2020-07-28:
Thanks Mushegh, So I can use the same demotrackinfo in my multitrack MediaSource configuration and It will support .h264 video .g711 audio frames. no need to change codeId,codacData for PCM audio frame in demo trackinfo kinesis will handle it am I right?

MushMal, 2020-07-28:
The content type as well as the track codec id are used in downstream processes - it depends what they expect. If you have an application that consumes and knows about its tracks then you can set this to anything you want.

Content type is specified as a concatenation of the particular types - something like this "video/h264,audio/aac" for h264 video and aac audio. For g711 I believe it's "audio/mulaw" but I am not entirely sure.

The codec id I am not entirely sure if there is a standardization for g711. For AAC we use "A_AAC".

amiya-elear, 2020-07-29:
Thanks Mushegh,
If I provide withcontenttype in my configuration then do I need to provide trackinfo? if yes then what will be my codecPrivate data for codecId "A_MS/ACM" PCMA audio()?
Below is my MediaSource configuration.
```
private static MyAudioVideoMediaSource createMyAudioVideoMediaSource() {
        TrackInfo[] trackInfos = new TrackInfo[2];
        trackInfos[0]=new TrackInfo(1,"V_MPEG4/ISO/AVC","VideoTrack",AVCC_EXTRA_DATA, MkvTrackInfoType.VIDEO);
        trackInfos[1]=new TrackInfo(2,"A_MS/ACM","AudioTrack",PCMA_EXTRA_DATA, MkvTrackInfoType.AUDIO);
        final MultiTrackMediaSourceConfiguration configuration =
                new MultiTrackMediaSourceConfiguration.Builder()
                        .withFps(25)
                        .withTrackInfoList(trackInfos)
                        .withRetentionPeriodInHours(1)
                        .withAbsoluteTimecode(ABSOLUTE_TIMECODES)
                        .build();
        final MyAudioVideoMediaSource mediaSource = new MyAudioVideoMediaSource(STREAM_NAME);
        mediaSource.configure(configuration);
        return mediaSource;
    }
```
If I provide only audio and video content type ".withContentType("video/h264,audio/araw")" not providing Trackinfo in my media source confguration I am getting a run time error as below.
```
Exception in thread "main" java.lang.NullPointerException
	at com.amazonaws.kinesisvideo.producer.StreamInfo.getTrackInfoCount(StreamInfo.java:367)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createKinesisVideoStream(Native Method)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createStream(NativeKinesisVideoProducerJni.java:327)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createStreamSync(NativeKinesisVideoProducerJni.java:353)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.registerMediaSource(NativeKinesisVideoClient.java:131)
	at buzz.xxxx.KinesisHlsDemoStream.main(KinesisHlsDemoStream.java:64)
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed converting stream info object. StatusCode: 0xd
	at buzz.xxxxxxx.KinesisHlsDemoStream.main(KinesisHlsDemoStream.java:70)
```
Can you please help, What I am doing wrong here?

MushMal, 2020-07-29:
You need to provide the track info. Normally, your media pipeline should provide you the CPD that you can use or you need to create it, depending on the codec configuration.

The native code core has an API to generate CPDs for different codecs. For example: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/mkvgen/src/MkvGenerator.c#L1704. You could potentially adapt it in your code.

There is more info available for the playback here: https://aws.amazon.com/kinesis/video-streams/faqs/

amiya-elear, 2020-07-29:
Thanks Mushegh.
.withContentType("video/h264,audio/mrlaw") in MultiTrackMediaSourceConfiguration. 
I am able to send audio and video frame but when I am  playback the video in kinesis I am getting 
 ERROR:The type of the media is not supported or could not be determined from the media codec ids: (track 1: V_MPEG4/ISO/AVC), (track 2: A_MS/ACM).

MushMal, 2020-07-29:
@MixMasterMitch could you comment?

MixMasterMitch, 2020-10-30:
Per https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_reader_GetHLSStreamingSessionURL.html 'audio/alaw' is not supported for playback via HLS. The only audio codec supported by the HLS protocol and KVS is AAC.

Per https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_reader_GetDASHStreamingSessionURL.html 'audio/alaw' can be used with MPEG-DASH. HOWEVER, most MPEG-DASH players will not support alaw, only AAC.