awslabs/amazon-kinesis-video-streams-producer-sdk-java #107: Aws kinesis video stream - audio track not working

my56, 2020-07-17:
I'm trying to stream video and audio in multitrack,I'm gathering audio as aac using java sound api, video as h264 using webcam api and for multitrack I'm referring to AudioVideoFrameSource, AudioVideoFileMediaSourceConfiguration and AudioVideoFileMediaSource class.Not dealing with the thread sleep between putting audio/video frames and the file related operations.

Here I'm facing audio and video sync issue,audio plays like a fast forwarding mode.

[AWS demo.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/4936032/AWS.demo.zip)

disa6302, 2020-07-17:
@my56 , 

Are you removing the THREAD_SLEEP() ?

my56, 2020-07-20:
@disa6302 Yes,we have removed thread sleep between audio/video frames.

closed 2020-11-08 by MushMal