awslabs/amazon-kinesis-video-streams-producer-sdk-java #109: Is there a way to know any active client using the generated HLS URL?

amiya-elear, 2020-07-23:
Hi Team,
Is there a way to know any active client using the HLS URL generated from a kinesis live video stream?

MushMal, 2020-07-23:
If I understand your question correctly, you are asking whether there is an active HLS session in progress that pulls from the URL generated earlier? I can't really think of any easy way. What's the scenario you are using?

amiya-elear, 2020-07-23:
Thanks, Mushega for your prompt response.
The scenario is I am doing a live video stream and I generated an HLS session URL.  If my generated HLS session URL is not using by an active client (any player not playing it) I want to stop the live stream.

MushMal, 2020-07-23:
I don't see an easy way of knowing post-factum. As you are controlling who you give the generated URL you could then ensure whether the playback is in progress from that entity I guess?

amiya-elear, 2020-07-23:
Thanks Mushegh, So there is no easy way to ensure my generated HLS URL using by any active client or not. We have to check playback is in progress from that entity or not. Do you have any documentation regarding this so that I can go through and check?

MushMal, 2020-07-23:
The place to look is the AWS documentation at https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/hls-playback.html

sharadrj, 2020-08-08:
Closing now. @amiya-elear, please reopen or start a new issue if you have further questions.

closed 2020-08-08 by sharadrj