awslabs/amazon-kinesis-video-streams-producer-sdk-java #110: KNIVideoProducer Java SDK - Unsatisfied Link Error

has253, 2020-07-27:
Hi Team,

I'm trying to create KVS video stream using Java SDK but getting the following error. The error says "java.lang.RuntimeException". I understand from a previous post that this exact same issue was raised earlier in February as well. (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/81) Please have a look and suggest a solution.
[image]

MushMal, 2020-07-27:
Could you try a new clean installation with the latest versions following the readme. It should be finding the binary for the later versions

has253, 2020-07-28:
I tried a few times but couldn't get through this error. Anyway, I am trying to create a new C++ based SDK but running into error with regards to cmake. I've followed all the instructions as listed here (https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-win.html) but I'm getting the error cmake: command not found. I manually tried copying cmake files downloaded from https://cmake.org/download/ to the source in mysys post which I ran into the following error.    

$ cmake ..
-- Building for: Visual Studio 16 2019
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
-- The Fortran compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_Fortran_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "C:/msys64/home/hunai/amazon-kinesis-video-streams-producer-c/build/CMakeFiles/CMakeOutput.log".
See also "C:/msys64/home/hunai/amazon-kinesis-video-streams-producer-c/build/CMakeFiles/CMakeError.log".

MushMal, 2020-07-28:
I am not sure what’s going on but you will need to clean install the prerequisites like the visual studio etc... take a look at our Travis CI script installing on Windows

has253, 2020-07-28:
Can you provide the URL to Travis CI script please?

MushMal, 2020-07-28:
The C++ repository is available here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp

The C repository is available here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-c

C++ wraps the C SDK, which, in turn, wraps the core.

There is a .travis.yml file in the root directory in all of these repositories.

hassanctech, 2020-10-30:
@kavyasoni Are you still seeing this issue?  If so can you please cut a new issue with your setup detailing the steps you've taken so far, which commit you're using,  I'm closing this issue for now due to no activity.

closed 2020-10-30 by hassanctech