awslabs/amazon-kinesis-video-streams-producer-sdk-java #111: PCM A-law Audio

amiya-elear, 2020-07-29:
Hi,

My camera generates H.264 1080p, 25fps and G.711 PCM A-law 8Khz, Mono stream.
When I stream Video Only content, I am able to record and playback using both HLS and DASH.

But when I add the audio track, while I am able to push the data to Kinesis, and generate the DASH URL. I am facing some problems during playback.
I get the following message from the video player in the console: "The type of the media is not supported or could not be determined from the media codec ids: (track 1: V_MPEG4/ISO/AVC), (track 2: A_MS/ACM)."

I suspect I am have an incorrect value for the codecId: G.711 PCM A-law, as well as I am quite unsure on how to use the codecPrivateData. I presume Kinesis is storing the data in an MKV file?

I have initialised the tracks as follows:

    private static final byte[] AVCC_EXTRA_DATA = {(byte) 0x01, (byte) 0x42, (byte) 0xc0, (byte) 0x28, (byte) 0xff,
            (byte) 0xe1, (byte) 0x00, (byte) 0x1a, (byte) 0x67, (byte) 0x42, (byte) 0xc0, (byte) 0x28, (byte) 0xdb,
            (byte) 0x02, (byte) 0x80, (byte) 0xf6, (byte) 0xc0, (byte) 0x5a, (byte) 0x80, (byte) 0x80, (byte) 0x80,
            (byte) 0xa0, (byte) 0x00, (byte) 0x00, (byte) 0x03, (byte) 0x00, (byte) 0x20, (byte) 0x00, (byte) 0x57,
            (byte) 0xe4, (byte) 0x01, (byte) 0xe3, (byte) 0x06, (byte) 0x5c, (byte) 0x01, (byte) 0x00, (byte) 0x04,
            (byte) 0x68, (byte) 0xca, (byte) 0x8c, (byte) 0xb2};
    
    private static final byte[] PCMA_EXTRA_DATA = {(byte) 0x11, (byte) 0x90}; //This is incorrect and I need guidance here.

    /**
     * Create a MediaSource based on local sample H.264 frames.
     *
     * @return a MediaSource backed by local H264 frame files
     */
    private static CocoAudioVideoMediaSource createCocoAudioVideoMediaSource() {
        TrackInfo[] trackInfos = new TrackInfo[2];
        trackInfos[0]=new TrackInfo(1,"V_MPEG4/ISO/AVC","VideoTrack",null, MkvTrackInfoType.VIDEO);
        trackInfos[1]=new TrackInfo(2,"A_MS/ACM","AudioTrack",PCMA_EXTRA_DATA, MkvTrackInfoType.AUDIO);
        final MultiTrackMediaSourceConfiguration configuration =
                new MultiTrackMediaSourceConfiguration.Builder()
                        .withTrackInfoList(trackInfos)
                        .withRetentionPeriodInHours(1)
                        .build();
        final CocoAudioVideoMediaSource mediaSource = new CocoAudioVideoMediaSource(STREAM_NAME);
        mediaSource.configure(configuration);
        System.out.println("media*****************"+mediaSource);
        return mediaSource;
    }

Could you kindly help me by providing me the correct values to use here?

Thanks.

MushMal, 2020-07-29:
You need to provide the CPD to the track info. You can generate it following the same pattern as in the native code. We don’t have the same utility functionality in Java unfortunately.

amiya-elear, 2020-07-31:
Hi Mushegh ,
Do we have any interface in kinesis Java SDK to generate CPD for PCM audio? How we are generating in C
https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/mkvgen/src/MkvGenerator.c#L1704

MushMal, 2020-07-31:
Unfortunately, this is not exposed through JNI. Though, the code is very straight forward to replicate in Java directly

amiya-elear, 2020-08-03:
Thanks, Mushmal.
I have generated the CPD for PCM using  mkvgenGeneratePcmCpd() for format  KVS_PCM_FORMAT_CODE_ALAW. After putting  audio and video frame into kinesis stream when I am playing back video in dash player

I am gettoing error: audioCodec (audio/mp4;codecs="alaw") is not supported.
```
[INFO] Page loaded
[INFO] Fetching data endpoint
[INFO] Data endpoint: https://b-xxxxx.kinesisvideo.us-west-2.amazonaws.com
[INFO] Fetching DASH Streaming Session URL
[INFO] DASH Streaming Session URL: https://xxxxxxxxxxxx
[INFO] Created DASH.js Player
[INFO] Starting playback
[INFO] Set player source
[ERROR] [403][Stream] audioCodec (audio/mp4;codecs="alaw") is not supported.
```
**where i am doing wrong**
```
**Code for MediaSource:**
 private static final byte[] AVCC_EXTRA_DATA = {(byte) 0x01, (byte) 0x42, (byte) 0xc0, (byte) 0x28, (byte) 0xff,
            (byte) 0xe1, (byte) 0x00, (byte) 0x1a, (byte) 0x67, (byte) 0x42, (byte) 0xc0, (byte) 0x28, (byte) 0xdb,
            (byte) 0x02, (byte) 0x80, (byte) 0xf6, (byte) 0xc0, (byte) 0x5a, (byte) 0x80, (byte) 0x80, (byte) 0x80,
            (byte) 0xa0, (byte) 0x00, (byte) 0x00, (byte) 0x03, (byte) 0x00, (byte) 0x20, (byte) 0x00, (byte) 0x57,
            (byte) 0xe4, (byte) 0x01, (byte) 0xe3, (byte) 0x06, (byte) 0x5c, (byte) 0x01, (byte) 0x00, (byte) 0x04,
            (byte) 0x68, (byte) 0xca, (byte) 0x8c, (byte) 0xb2};
    
    private static final byte[] PCMA_EXTRA_DATA_2 = {
    (byte)0x06, (byte)0x00, (byte)0x01, (byte)0x00, (byte)0x40, (byte)0x1f, (byte)0x00, (byte)0x00, (byte)0xe8,
            (byte)0x03, (byte)0x00, (byte)0x00, (byte)0x01, (byte)0x00,
            (byte)0x00, (byte)0x00, (byte)0x00, (byte)0x00};

    /**
     * Create a MediaSource based on local sample H.264 frames.
     *
     * @return a MediaSource backed by local H264 frame files
     */
    private static CocoAudioVideoMediaSource createCocoAudioVideoMediaSource() {
        TrackInfo[] trackInfos = new TrackInfo[2];
       trackInfos[0]=new TrackInfo(1,"V_MPEG4/ISO/AVC","VideoTrack",AVCC_EXTRA_DATA, MkvTrackInfoType.VIDEO);
       trackInfos[1]=new TrackInfo(2,"A_MS/ACM","AudioTrack",PCMA_EXTRA_DATA_2, MkvTrackInfoType.AUDIO);
        final MultiTrackMediaSourceConfiguration configuration =
                new MultiTrackMediaSourceConfiguration.Builder()
                        .withTrackInfoList(trackInfos)
                        .withRetentionPeriodInHours(1)
                        .build();
        final CocoAudioVideoMediaSource mediaSource = new CocoAudioVideoMediaSource(STREAM_NAME);
        mediaSource.configure(configuration);
        return mediaSource;
    }
}
```
**Code for put frame into Kinsis**
```
private void VideoCallback(int index, int flags, long decodingTs, long presentationTs, long duration,
                               @Nonnull ByteBuffer data, long trackId){
        try {
            KinesisVideoFrame frame = new KinesisVideoFrame(
                    index,
                    flags,
                    decodingTs,
                    presentationTs,
                    duration,
                    data,
                    trackId);
            PutCocoframe (frame);
        } catch (KinesisVideoException e) {
            e.printStackTrace();
        }
    }

    private void AudioCallback(int index, int flags, long decodingTs, long presentationTs, long duration,
                               @Nonnull ByteBuffer data, long trackId){
        try {
            KinesisVideoFrame frame = new KinesisVideoFrame(
                    index,
                    flags,
                    decodingTs,
                    presentationTs,
                    duration,
                    data,
                    trackId);
            PutCocoframe(frame);
        } catch (KinesisVideoException e) {
            e.printStackTrace();
        }
    }

    private synchronized void PutCocoframe(KinesisVideoFrame frame ) throws KinesisVideoException {
        Timestamp timestamp = new Timestamp(System.currentTimeMillis());
        listener.onFrameDataAvailable(frame);
    }
}
```
Can you help me to identify the issue?

MushMal, 2020-08-03:
Apparently, the console playback doesn't support a-law. I am assigning this to @MixMasterMitch for further comment as to what's supported.

MixMasterMitch, 2020-08-03:
Yes, browsers do not support any sort of PCM based audio via HLS/DASH. AAC is the only codec support by both KVS and browsers.

closed 2020-08-11 by amiya-elear