awslabs/amazon-kinesis-video-streams-producer-sdk-java #112: Failed to put a frame into the stream. StatusCode: "0x30000005"

amiya-elear, 2020-08-04:
Hi,
I am getting Failed to put a frame into the stream. StatusCode: "0x30000005" error while transmitting audio and video frame together using custom media sources.
Below I provided a link for custom AudioVideoFrameSource and MediaSourceConfiguration. Now I am reading from a file and sending frames into kinesis for a demo to test audio and video playing properly or not.
************************************
Here is my custom AudioVideoFrameSource
https://drive.google.com/file/d/1fmfQfJo_ew28q0z08my1mxSBKqjfhkTX/view?usp=sharing

Here is my DemoApp
https://drive.google.com/file/d/1Fypmsmz692iGTHGWUdvC8qXbwn-xFtL-/view?usp=sharing

**************************************************************************************
while I am trnsmiting only Video using video track I am able to send video and play it but when I am sending audio and video together its falling with "Failed to put a frame into the stream. StatusCode: "0x30000005"
can you help me why I am getting StatusCode: "0x30000005" while transmiting audio and video together?

In below link detail kinesis log
https://drive.google.com/file/d/1U0E7eHSwQ3b3ZA2Gbtx1Sqqku6mBQ-fd/view?usp=sharing
 
Thanks.

MushMal, 2020-08-04:
Here is the list of status codes the SDK returns: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html

STATUS_CONTENT_VIEW_INVALID_TIMESTAMP indicates that the frames are not monotonically increasing.

The older version of the SDK has a strict monotonically increasing enforcement. Could you please specify a small duration for both of the tracks - something like 2 millis. Your audio tracks have 0 duration.

amiya-elear, 2020-08-04:
Thanks, Mushegh,
I have checked the status code but in my case, I am increasing my PTS in a fixed number for audio and video so it should not provide any issue, correct me if I wrong.

I have provided duration 2* HUNDREDS_OF_NANOS_IN_A_MILLISECOND but still, I am getting the same error.

Kinesis Logs::::::::::::::::::::::
```
DEBUG / KinesisVideo: PutFrame index: 0, pts: 670100, dts: 670100, duration: 20000, keyFrame: true, track: 1
DEBUG / KinesisVideo: Kinesis Video client and stream metrics
	>> Overall storage size: 1073741824
	>> Available storage size: 1073741824
	>> Allocated storage size: 0
	>> Total view allocation size: 48072
	>> Total streams frame rate: 25
	>> Total streams transfer rate: 2000000
	>> Current view duration: 0
	>> Overall view duration: 0
	>> Current view size: 0
	>> Overall view size: 0
	>> Current frame rate: 25.0
	>> Current transfer rate: 2000000
DEBUG / KinesisVideo: PutFrame index: 0, pts: 1280100, dts: 1280100, duration: 20000, keyFrame: false, track: 2
DEBUG / KinesisVideo: PutFrame index: 1, pts: 1340100, dts: 1340100, duration: 20000, keyFrame: false, track: 1
DEBUG / KinesisVideo: PutFrame index: 2, pts: 2010100, dts: 2010100, duration: 20000, keyFrame: false, track: 1
DEBUG / KinesisVideo: PutFrame index: 1, pts: 2560100, dts: 2560100, duration: 20000, keyFrame: false, track: 2
DEBUG / KinesisVideo: PutFrame index: 3, pts: 2680100, dts: 2680100, duration: 20000, keyFrame: false, track: 1
DEBUG / KinesisVideo: PutFrame index: 2, pts: 3840100, dts: 3840100, duration: 20000, keyFrame: false, track: 2
DEBUG / KinesisVideo: PutFrame index: 4, pts: 3350100, dts: 3350100, duration: 20000, keyFrame: false, track: 1
com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream. StatusCode: 0x30000005
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putKinesisVideoFrame(Native Method)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putFrame(NativeKinesisVideoProducerJni.java:491)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerStream.putFrame(NativeKinesisVideoProducerStream.java:271)
	at com.amazonaws.kinesisvideo.internal.mediasource.ProducerStreamSink.onFrame(ProducerStreamSink.java:36)
	at com.amazonaws.kinesisvideo.internal.mediasource.DefaultOnStreamDataAvailable.onFrameDataAvailable(DefaultOnStreamDataAvailable.java:21)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource.PutCocoframe(CocoAudioVideoFrameSource.java:314)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource.VideoCallback(CocoAudioVideoFrameSource.java:284)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource.access$200(CocoAudioVideoFrameSource.java:29)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource$1.run(CocoAudioVideoFrameSource.java:127)
	at java.lang.Thread.run(Thread.java:748)
DEBUG / KinesisVideo: PutFrame index: 5, pts: 4020100, dts: 4020100, duration: 20000, keyFrame: false, track: 1
DEBUG / KinesisVideo: PutFrame index: 3, pts: 5120100, dts: 5120100, duration: 20000, keyFrame: false, track: 2
DEBUG / KinesisVideo: PutFrame index: 6, pts: 4690100, dts: 4690100, duration: 20000, keyFrame: false, track: 1
com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream. StatusCode: 0x30000005
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putKinesisVideoFrame(Native Method)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putFrame(NativeKinesisVideoProducerJni.java:491)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerStream.putFrame(NativeKinesisVideoProducerStream.java:271)
	at com.amazonaws.kinesisvideo.internal.mediasource.ProducerStreamSink.onFrame(ProducerStreamSink.java:36)
	at com.amazonaws.kinesisvideo.internal.mediasource.DefaultOnStreamDataAvailable.onFrameDataAvailable(DefaultOnStreamDataAvailable.java:21)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource.PutCocoframe(CocoAudioVideoFrameSource.java:314)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource.VideoCallback(CocoAudioVideoFrameSource.java:284)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource.access$200(CocoAudioVideoFrameSource.java:29)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource$1.run(CocoAudioVideoFrameSource.java:127)
	at java.lang.Thread.run(Thread.java:748)
DEBUG / KinesisVideo: PutFrame index: 7, pts: 5360100, dts: 5360100, duration: 20000, keyFrame: false, track: 1
DEBUG / KinesisVideo: PutFrame index: 4, pts: 6400100, dts: 6400100, duration: 20000, keyFrame: false, track: 2
DEBUG / KinesisVideo: PutFrame index: 8, pts: 6030100, dts: 6030100, duration: 20000, keyFrame: false, track: 1
com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream. StatusCode: 0x30000005
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putKinesisVideoFrame(Native Method)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putFrame(NativeKinesisVideoProducerJni.java:491)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerStream.putFrame(NativeKinesisVideoProducerStream.java:271)
	at com.amazonaws.kinesisvideo.internal.mediasource.ProducerStreamSink.onFrame(ProducerStreamSink.java:36)
	at com.amazonaws.kinesisvideo.internal.mediasource.DefaultOnStreamDataAvailable.onFrameDataAvailable(DefaultOnStreamDataAvailable.java:21)
[... cut ...]
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource.access$200(CocoAudioVideoFrameSource.java:29)
	at buzz.getcoco.alexa.CocoAudioVideoFrameSource$1.run(CocoAudioVideoFrameSource.java:127)
	at java.lang.Thread.run(Thread.java:748)
DEBUG / KinesisVideo: PutFrame index: 10, pts: 7370100, dts: 7370100, duration: 20000, keyFrame: false, track: 1
DEBUG / KinesisVideo: PutFrame index: 6, pts: 8960100, dts: 8960100, duration: 20000, keyFrame: false, track: 2
DEBUG / KinesisVideo: PutFrame index: 11, pts: 8040100, dts: 8040100, duration: 20000, keyFrame: false, track: 1
```

MushMal, 2020-08-04:
This is odd, the first instance of the error happens on TS: 3350100 which is greater than the previous frame TS + duration of the same track: 2680100. It seems that the track1 frame of index 4 interferes with frame index 2 of track 2. 

Need to understand why. You seem to specify the correct track info and proper frame order

MushMal, 2020-08-04:
I also assume you are using USE_FRAME_TIMECODES (default in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/8d7faf74297e9bb5b5809b792b1656ee66d6998a/src/main/java/com/amazonaws/kinesisvideo/internal/mediasource/multitrack/MultiTrackMediaSource.java#L63)

amiya-elear, 2020-08-04:
Yes, I am using the same configuration in my code
here is my CocoAudioVideoMediaSource
https://drive.google.com/file/d/1-3pxAGtDYd5O0roiz2-imWJWEkhscbDo/view?usp=sharing

amiya-elear, 2020-08-04:
Can you help me to find where I am doing wrong? So that I can resolve this issue.

MushMal, 2020-08-04:
Absolutely, let me look deeper into this. From what I have seen so far, it should work just fine. Not sure why the content view checks the other tracks timestamps

amiya-elear, 2020-08-04:
Thanks Mushegh.

MushMal, 2020-08-04:
Somehow, it looks as if though the frame order mode is not properly set. Could you please print out the value of https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/31b1b11dd64f332f1167d15c12bfc5699d67de4c/src/main/java/com/amazonaws/kinesisvideo/producer/StreamInfo.java#L467
It should be 1, 2 or 3

amiya-elear, 2020-08-04:
So  I have to create an object of StreamInfo and then print the value of getFrameOrderMode() method before each and every put frame call?

MushMal, 2020-08-04:
No, the media source would eventually end up with this object. The native code then will call this API directly once. Just add a logging statement to print the value in the getter.

amiya-elear, 2020-08-04:
so I am trying to print like this 
System.out.println("getFrameOrderMode"+ mediaSource.getStreamInfo().getFrameOrderMode()); but its showing not object in this name.

MushMal, 2020-08-04:
You can simply modify the code in the getter and add System.out.println("order mode: " + mFrameOrderMode);

MushMal, 2020-08-04:
The mediaSource.getStreamInfo() should also return the stream info object. What exactly does it print?

amiya-elear, 2020-08-04:
mediaSource.getStreamInfo() printing link com.amazonaws.kinesisvideo.producer.StreamInfo@6267c3bb
and before that in debug log I got
DEBUG / KinesisVideo: describe stream result: {StreamInfo: {StreamName: coco-stream-1,StreamARN: arn:aws:kinesisvideo:us-west-2:408257412428:stream/coco-stream-1/1596510288069,KmsKeyId: arn:aws:kms:us-west-2:408257412428:alias/aws/kinesisvideo,Version: Tlf7MApZPbXw6uPpXAU7,Status: ACTIVE,CreationTime: Tue Aug 04 08:34:48 IST 2020,DataRetentionInHours: 24}}

amiya-elear, 2020-08-04:
Am I doing anything wrong while printing? I think describe stream also provied the same streaminfo.

MushMal, 2020-08-04:
Describe call result will not have any of this info as this is purely producer side setting.

Not sure why you can't get mediaSource.GetStreamInfo. getFrameOrderMode printed out. You could also simply add debug lines to the StreamInfo.java getFrameOrderMdoe to print it out. Alternatively, you can put a breakpoint or something.

What I am seeing from the logs is that the frame order mode is set to pass-through which will cause this error as it won't sort the frame timestamps from different tracks.

amiya-elear, 2020-08-05:
I am getting the value of getFrameOrderMdoe as 3.

MushMal, 2020-08-05:
I am still unsure what's going on. Your logs indicate an extremely high frame rate observed - upto 74000 fps - not sure why.

The logs indicate successful streaming for the duration of the log but there are also failures with the same error 0x30000005.

Could you try to specify 0 duration for the frames in order to ensure the frames don't overlap from different tracks.

amiya-elear, 2020-08-05:
Sure, I have tried with 0 frame duration now it's not giving any error but while playing live playback, it's not playing the video. 
it's showing below message. 
Download your Kinesis Video Stream Producer SDK
Use the Kinesis Video Streams Producer SDK to set up your connected device as a producer for this video stream. Also verify that

But when I am trying to play as on-demand playback using browser by providing server TS it's playing the video without audio because the browser doesn't support Alw formatted audio.

MushMal, 2020-08-05:
I really don't see any issues with the logs. The only reason the playback doesn't work is due to the codec of the audio. KVS provides an ability to generate MPEG-DASH stream which could contain PCM.

Could you please go ahead and resolve this issue as the root issue has been resolved for you

amiya-elear, 2020-08-05:
Thanks, Mushegh.
 The codec of audio is not correct or the player does not support PCM audio to playback?
Can you please help with some player where I can do live playback  MPEG-DASH session URL.

MushMal, 2020-08-05:
Apologies, I meant the console playback doesn’t support the pcm playback. Regarding the mpeg-dash.. most platforms have a player that support dash playback. Which platform are you targeting?

MushMal, 2020-08-05:
@MixMasterMitch might have a good idea what’s available for dash playback on which platform

amiya-elear, 2020-08-05:
I want to play this video in my Chromecast.

amiya-elear, 2020-08-10:
Any update on this not able to play KVS generated HLS URL in Chromecast?

closed 2020-08-11 by amiya-elear

wlinInspire, 2020-11-15:
> Here is the list of status codes the SDK returns: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html
>
> STATUS_CONTENT_VIEW_INVALID_TIMESTAMP indicates that the frames [...]

@MushMal Is `0 duration` equivalent to `fragment-duration=0` in the gstreamer plugin

MushMal, 2020-11-16:
@wlinInspire fragment-duration specifies the duration of the fragment if the SDK is configured to do fragmentation. 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L452

By default, the SDK is fragmenting every time the media pipeline produces a key-frame. For most video streaming cases the encoder should be configured to produce an Idr frame every 2-8 seconds which is the GoP size (Group-of-pictures) and the Idr frame is marked as a key-frame which then drives the SDK to do the fragmentation.

In some cases - such as Audio stream only, there is no concept of a fragment natively. So, the SDK can be configured to NOT produce a fragment on every key-frame but rather on any key-frame after "fragment-duration" has elapsed. This way, the audio media pipeline can be configured to produce every frame as a key-frame and the fragmentation will happen after fragment-duration.

wlinInspire, 2020-11-16:
@MushMal Thanks for the explanation!