awslabs/amazon-kinesis-video-streams-producer-sdk-java #113: Failed to put a frame into the stream. StatusCode: 0x52000085

amiya-elear, 2020-08-08:
Hi Team,
I have a problem with sending audio and video to KVS, I am using MultiTrackMediaSourceConfiguration to send my audio and video frame into kinesis. And I'm getting this

```
DEBUG / KinesisVideo: PutFrame index: 236, pts: 396405542200, dts: 396405542200, duration: 0, keyFrame: false, track: 2
com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream. StatusCode: 0x52000085
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putKinesisVideoFrame(Native Method)

************************************producer timestamp***********************2020-08-08 13:24:44.317
DEBUG / KinesisVideo: PutFrame index: 244, pts: 396410150100, dts: 396410150100, duration: 0, keyFrame: false, track: 2
to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396402086100 presentation timestamp: 396402086100
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396402662100 presentation timestamp: 396402662100
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396403238100 presentation timestamp: 396403238100
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396403814200 presentation timestamp: 396403814200
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396404390200 presentation timestamp: 396404390200
2020-08-08 07:54:42 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396404966100 presentation timestamp: 396404966100
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396405542200 presentation timestamp: 396405542200
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396406118200 presentation timestamp: 396406118200
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396406694100 presentation timestamp: 396406694100
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396407270100 presentation timestamp: 396407270100
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396407846100 presentation timestamp: 396407846100
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396408422100 presentation timestamp: 396408422100
2020-08-08 07:54:43 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:44 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396408998200 presentation timestamp: 396408998200
2020-08-08 07:54:44 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085
2020-08-08 07:54:44 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 396409574100 presentation timestamp: 396409574100
2020-08-08 07:54:44 WARN    putKinesisVideoFrame(): Failed to put a frame with status code 0x52000085

```
I was trying to find this status code 0x52000085 , but I can't find any information in AWS Documentation.

can you help me to resolve the issue.

MushMal, 2020-08-08:
Yeah, unfortunately the status codes documentation has not been updated to reflect the latest code as the code is constantly being updated.

This status code is defined in https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L182

This indicates that you have a large drift between the audio and video track frames. When using multi-track solution, the av-sync logic will try to cache the audio frames and release them into the core in sync with video.

amiya-elear, 2020-08-09:
Thanks Mushegh,
 
Is it because of the large PTS difference between audio and video frames?

MushMal, 2020-08-09:
Oh, sorry, yes, that’s what I tried to imply. Sometimes that happens when the audio pipeline keeps producing frames and the video pipeline stalls but there are also cases when the audio and video pipelines are not on the same timeline

amiya-elear, 2020-08-09:
Thanks, Mushegh, So Can you please guide me what I have to do to resolve this issue? My camera is providing me PTS,DTS for audio, video frames I am just mapping those PTS, DTS, and putting frames into kinesis using audio and video track. where I am doing wrong?

MushMal, 2020-08-09:
I would recommend first if all dumping out the timestamps of the frames for both tracks and trying to understand which case you are hitting. If it’s the case with the stale video encoder then perhaps you will need to consult with your encoder manufacturer to understand how to avoid it. If it’s the case of different timelines then you can perhaps capture the diff in the timeline and try to apply the diff to the frames before putting them into the KVS to bring them to the same timeline.

amiya-elear, 2020-08-10:
Thanks, Mushmal.

MushMal, 2020-08-10:
Resolving this issue for now. Feel free to open an issue when you have a question

closed 2020-08-10 by MushMal