awslabs/amazon-kinesis-video-streams-producer-sdk-java #114: calculate video CPD for  V_MPEG/ISO/AVC and contain H.264 encoded media.

amiya-elear, 2020-08-11:
Hi Team,

Is there any way to calculate video CPD for  V_MPEG/ISO/AVC and contain H.264 encoded media.
How we are calculating for audio here: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/mkvgen/src/MkvGenerator.c#L1704.
Thakns.

MushMal, 2020-08-11:
It's possible to generate the CPD for h264 (in h264 it's SPS and PPS) if you know all of the details from the encoder.

Generally, your encoder should have a way to communicate the SPS and PPS to the clients. Some of the media pipelines have a backchannel to get the CPD while others would prepend the SPS and PPS in Annex-B format to the I-frame.

We have code that automatically extracts it from Annex-B format stream if it's prepended to the I-frame, no CPD had been specified yet and the adaptation flags have Annex-B -> AvCC format adaptation.

https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/mkvgen/src/MkvGenerator.c#L235

Please let us know if you have further questions and if not please go ahead and resolve the issue

amiya-elear, 2020-08-12:
Is KVS java SDK provide a similar function like mkvgenGeneratePcmCpd() to generate CPD h264 video frame? So that how we are providing required data to generate CPD for PCM audio in similar way we can create CPD for h264  video frames by providing required data?

MushMal, 2020-08-12:
Per my answer above - you can’t generate the CPD for h264. Perhaps I am not understanding your question? Could you clarify?

amiya-elear, 2020-08-12:
Thanks Mushmal. I got it.

closed 2020-08-12 by MushMal