awslabs/amazon-kinesis-video-streams-producer-sdk-java #115: Not able to play KVS generated HLS and MPEG-DASH URL in Chromecast?

amiya-elear, 2020-08-11:
Hi Team,

I have generated an HLS and MPEG-DASH URL(fragemented_mp4) from my KVS camera stream, Chromecast is not playing this HLS and MPEG-DASHURL.

Thanks.

MushMal, 2020-08-11:
Could you please provide more details? Is this a decoding issue or an actual HLS/MPEG-DASH issue? Does the HLS player in the browser play the content? 

There is a section in the docs going over the troubleshooting steps: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/troubleshooting.html#troubleshooting-hls

More on DASH: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/dash-playback.html

amiya-elear, 2020-08-12:
Thanks Mushegh
I am not sure Is this a decoding issue or an actual HLS/MPEG-DASH issue. How can I check it is a decoding issue or not? yes in browser player I am able to play in dash, vlc, shaka but not playing in Chromecast.

MixMasterMitch, 2020-08-12:
Per [Chromecast Media Docs](https://developers.google.com/cast/docs/media), H.264 Baseline@4.0 is not explicitly listed as being supported. Can you try a different H.264 level/profile?

Can you also validate that Chromecast does not support your specific codec using the snipped from the docs?
```
   const context = cast.framework.CastReceiverContext.getInstance();
   context.canDisplayType('video/mp4', 'avc1.42c028');
```

amiya-elear, 2020-08-19:
Thanks for your response,
Is there any way to change the video profile/level while generating HLS or DASH URL? or while putting frame into kinesis?

MushMal, 2020-08-19:
KVS does not transcode the stream. You need to change the encoder settings on the device.

amiya-elear, 2020-08-19:
Thanks Mushegh, I am streaming h264 video frames into kinesis and generating HLS URL with container format fragemented_mp4 it's playing in AWS console as well as in browser with video profile baseline, level 3.0. This video profile supported by Chromecast as well, but when I am playing the HLS URL In Chromecast it's not playing.

MushMal, 2020-08-19:
Looks like @MixMasterMitch linked the documentation for Chromecast support. If it's not supported there isn't much can be done but to either transcode on the client or change the encoding at the source

amiya-elear, 2020-08-20:
Not sure why I am unable to play HLS URL in any player while generating HLS URL with container format mpeg_4. while generating container format with fragmented mp4 I am able to play in all players except Chromecast. but if I am able to play HLS URL with container format mpeg_4 in browser I am able to play in Chromecast as well.

closed 2020-09-08 by amiya-elear