awslabs/amazon-kinesis-video-streams-producer-sdk-java #116: [QUESTION] INVALID_MKV_DATA","ErrorId":4006 With encoded mkv generated with webms

Nipunsnapwiz, 2020-08-12:
Hi, I am using ffmpeg to generate mkvs from pre-recorded webm files recorded with Webrtc media recorder. I am using following command to convert:

ffmpeg -i <input path> -c:v libx264 -c:a aac <output path>

conversion is successful and output is playable in vlc but I am getting invalid mkv data error when using PUTMEDIA API to insert it as fragment to KVS. I upload mkv to S3 which is picked up by a lambda to inject in stream.
here is my PUT MEDIA CODE

            S3Object videoObject = s3.getObject(new GetObjectRequest(bucket, key));
            String contentType = videoObject.getObjectMetadata().getContentType();
//            String producerTimestamp = videoObject.getObjectMetadata().getUserMetaDataOf(System.getenv("PRODUCER_START_TIMESTAMP_KEY"));
            String producerTimestamp = videoObject.getObjectMetadata().getUserMetaDataOf("producer_start_timestamp");
            context.getLogger().log("Producer timestamp: " + producerTimestamp);
            Date producerStartTimestamp;
            try {
                producerStartTimestamp = new Date(Long.parseLong(producerTimestamp));
            } catch(Exception e) {
                System.out.println(e.getMessage());
                producerStartTimestamp = Date.from(Instant.now());
            }
            context.getLogger().log("Producer timestamp (date): " + producerStartTimestamp + "\n");
            context.getLogger().log("Content type: " + contentType + "\n");
            context.getLogger().log("Object key: " + key + "\n");

            String fileName = key.substring(key.lastIndexOf('/')+1, key.length());
            String tempWriteLocation = "/tmp/" + fileName;

            InputStream is = videoObject.getObjectContent();
            BufferedOutputStream outputStream = new BufferedOutputStream(new FileOutputStream(tempWriteLocation));
            int totalSize = 0;
            int bytesRead;
            byte[] content = new byte[1024 * 1014];
            while ((bytesRead = is.read(content)) != -1) {
//                context.getLogger().log(String.format("%d bytes read from S3 input stream\n", bytesRead));
                outputStream.write(content, 0, bytesRead);
                totalSize += bytesRead;
            }
            context.getLogger().log("Total size of file in bytes: " + totalSize + "\n");
            outputStream.close();

            try {
                streamToKVS(tempWriteLocation, streamName, producerStartTimestamp);
                return key;
            } catch (Exception e) {
                e.printStackTrace();
                context.getLogger().log(e.getMessage());
                return key;
            }

Can you tell me what I am doing wrong here as it also failed with mkv provided in java demo app also so I this there is some issue with PutMedia code instead of MKV file.

MushMal, 2020-11-03:
@Nipunsnapwiz have you done any further debugging? Can you attach the generated MKV here? NOTE: Not all types of MKV can be streamed.

MushMal, 2020-11-08:
Closing due to inactivity

closed 2020-11-08 by MushMal