awslabs/amazon-kinesis-video-streams-producer-sdk-java #117: video and audio resolution configuration not working

my56, 2020-08-14:
Not able to configure video resolution ,when I provide configuration as [176x144], [320x240] or [640x480] in aws configuration ,and view video in aws console it shows [640*480] as default for all resolution

below is the code used to set video configuration:

final AudioVideoCameraMediaSourceConfiguration configuration = new AudioVideoCameraMediaSourceConfiguration.AudioVideoBuilder()
				.withContentType(null)
				.withFps(FPS_50)
				.withRetentionPeriodInHours(RETENTION_ONE_HOUR)
				.withNalAdaptationFlag(StreamInfo.NalAdaptationFlags.NAL_ADAPTATION_ANNEXB_NALS)
				 .withCameraId("0")
				.withIsEncoderHardwareAccelerated(false)
				 .withEncodingMimeType(null)
				 .withEncodingBitRate(2000000)
				 .withHorizontalResolution(320)
				 .withVerticalResolution(240)
				 .withCodecPrivateData(cpd)
				.withAbsoluteTimecode(ABSOLUTE_TIMECODES)
				.withTrackInfoList(DemoTrackInfos.createTrackInfoList())
				.build();
		final AudioVideoCameraMediaSource mediaSource = new AudioVideoCameraMediaSource(StudentStreamName);
		mediaSource.setupWebCam(webcam);
		mediaSource.configure(configuration);

		return mediaSource;

	
And i am using webcam api to stream video,here i have also set configuration in webcam as below:
		if (!webcam.isOpen()) {
		webcam.setViewSize(new Dimension(320, 240));
		}
		webcam.open(true);
		
And for audio configuration if i set sample rate as 44100 in java sound api and stream audio,when i download from console and view in ffmpeg it shows sample rate as 48kb/s 

Please let us know where actual configuration for video and audio is happening in aws producer library for default streaming

hassanctech, 2020-08-25:
When you try to specify 320x240 as shown above and the console shows 640x480 resolution -- does the console actually show any video or is only the metadata populated?

hassanctech, 2020-08-25:
Also, is there a way you can verify that your media pipeline is actually producing frames with the dimensions that you specified?

my56, 2020-09-01:
@hassanctech ,
we are able view video and download video from aws console

lherman-cs, 2020-09-01:
@my56 as @hassanctech suggested, could you check if your media pipeline producing frames with the right dimensions?

my56, 2020-09-02:
Could you please suggest how to check media pipeline.Do we have any methods in aws producer java sdk as i am not using gstreamer to stream audio and video.
Could you please elaborate on that ,and could you specify any resource related to that.

hassanctech, 2020-09-22:
It looks like you are starting a stream from your webcam, is there a player like VLC you can play the stream in and verify the resolution, make sure to start it up exactly the same way as when sending it to KVS.

Did you adapt the AudioVideoFileMediaSourceConfiguration/AudioVideoFileMediaSource to use you camera? If so can you share the actual changes you made there?  Also does it work for you if you try the sample as is, can you see KVS showing the same resolution as what's in the sample?

Lastly just want to confirm the video you're able to view and download from the console it's the same video you're producing correct? And if you play that downloaded video in VLC for example does it report the same resolution as KVS?  We don't actually have any transcoding capabilities so it shouldn't be possible for the frame size to change.

hassanctech, 2020-10-30:
Closing due to inactivity.  Please open another issue if you require further assistance.

closed 2020-10-30 by hassanctech