awslabs/amazon-kinesis-video-streams-producer-sdk-java #118: how to check persisted FragmentAckType in kinesis MediaSource stream

amiya-elear, 2020-08-24:
Hi Team,

 I want to create a live dash/hls URL form my kinesis video stream. For that, according to aws doc we have wait until we will get a persisted FragmentAckType from the kinesis video stream. so now I am providing delay for 2 to 3 secs then I am calling getDashSessiomUrl. instead of delay, I want to wait for persisted FragmentAckType. how can I check for persisted FragmentAckType in Mediasource stream?
I checked in putmedia API, while strat the stream using putmedia API we can pass the response handler and check the ack type in a handler but for media source API how we can do that? or is there ay better way to handle it MediaSource stream?

```
final AmazonKinesisVideoArchivedMedia getDashurl = AmazonKinesisVideoArchivedMediaClient.builder()
                    .withCredentials(creds)
                    .withEndpointConfiguration(new AwsClientBuilder.EndpointConfiguration(dataEndpoint, DEFAULT_REGION))
                    .build();

            GetDASHStreamingSessionURLRequest req = new GetDASHStreamingSessionURLRequest()
                    .withDisplayFragmentTimestamp("NEVER")
                    .withStreamName(stream_name)
                    .withPlaybackMode("LIVE")
                    .withExpires(700);

            System.out.println("req****" + req.toString());
            TimeUnit.SECONDS.sleep(3);

            Dashurl = getDashurl.getDASHStreamingSessionURL(req).getDASHStreamingSessionURL();
```

MushMal, 2020-08-24:
MPEG-DASH or HLS playback is based on a fragments which have been persisted and indexed. 

While I am not sure I understand what your use case is and why you would want to "link" the producer and consumer side, you can take a look at https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/d490f31c692b1b2e989c58554ef00890ee62b515/src/main/java/com/amazonaws/kinesisvideo/producer/StreamCallbacks.java#L41 which is implemented in the default callbacks as NOOP: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/d490f31c692b1b2e989c58554ef00890ee62b515/src/main/java/com/amazonaws/kinesisvideo/streaming/DefaultStreamCallbacks.java#L26

amiya-elear, 2020-08-24:
Thanks, Mushegh,
Is there any way I can check for persisted fragmentAck in media source stream before calling getDASHStreamingSessionUR? Whenever I am trying to get Dash URL for the live streaming, if my stream does not have persisted data it's falling to provide me dash URL, throwing an exception . so before getting dash URL I want to check for persisted data then I want to call getDASHStreamingSessionUR.

Is there any better way to handle it?

MushMal, 2020-08-24:
Well, if you are on the same device then you could actually control when to issue getDASHStreamingSessionURI as you are the one that starts and stops producing frames.

In a more general case, you might do some polling with ListFragment API before issuing HSL or DASH - slightly higher TPS APIs.

Unfortunately, currently we don't have a notification system which is something we are working towards.

P.S. Alternatively, you could have some back channel from the producer to the consumer to kick off when the producer starts producing and retrieves it's first persisted ACK

amiya-elear, 2020-08-25:
So from the producer side how can I check for persisted ack ? after creating a media source I have to do just MediaSource.start() to produce data then how can check for persisted ACK from this stream?

MushMal, 2020-08-25:
You could derive from DefaultStreamCallbacks and handle the fragment acks and in your application you could supply the derived class.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/7d817e66670c6e807f35c9ed39ce61b2468648ab/src/main/java/com/amazonaws/kinesisvideo/java/client/KinesisVideoJavaClientFactory.java#L163