awslabs/amazon-kinesis-video-streams-producer-sdk-java #119: How long after no data is published does the connection close?

abdulsiddiqi, 2020-08-24:
And after a connection is closed, do I need to manually re-open the connection associated with the mediaSourceSink I've persisted or will it automatically re-open the connection if I publish a frame using the onFrame method?

MushMal, 2020-08-24:
The backend closes the connection after 30 seconds of inactivity. The latest producer SDK should handle the reconnect automatically once you push frames into the media source sink

closed 2020-08-24 by hassanctech

hassanctech, 2020-08-24:
Here is a link to documentation that advertises those numbers (and more):  https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_PutMedia.html

The relevant snippet:
"If no data is received within the 30 seconds, Kinesis Video Streams closes the request."