awslabs/amazon-kinesis-video-streams-producer-sdk-java #120: bufferDurationOverflowPressure callback fired

abdulsiddiqi, 2020-08-31:
For long running video streams this callback is getting fired along with the frame dropped ones too. I get the buffer might be getting too full which is causing bufferDurationOverflowPressure to be fired but why is it taking so long to get fired?

abdulsiddiqi, 2020-08-31:
We are getting the following message: "Buffer overflow pressure with remainDuration 9620000"

Some attributes from the StreamInfo structure are listed below:
bufferDuration = 17 seconds
replayDuration = 8.5 seconds
connectionStaleness duration = 9 seconds
timecodeScale = 1 second
maxLatency = 14 seconds
avgBandwidth = 2 * 1024 * 1024

lherman-cs, 2020-08-31:
@abdulsiddiqi As of now, `bufferDurationOverflowPressure` callback is called when your stream's buffer duration reaches 95% of its capacity, https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/1988f99fe77a3b37e9ed33643c20190a5c2e1689/src/client/src/Stream.c#L930-L946.

As the frames dropped, do you have multiple streams? If yes, I would also check the content storage size as all streams will share the same content storage, so it's possible that your content storage size is too small and causes the frames dropped earlier than expected. You can also find more details about this topic here, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/6c7d82eeaa3657709c13f0d36b0fabc3782b3cd8/docs/buffering.md.

abdulsiddiqi, 2020-09-01:
Is the content store part of the JVM heap or separate from it? The issue seems to be happening after streams have been running for a few hours (6 hours or more). Is it possible the client isn't doing a good job at clearing up previous frames over time?

lherman-cs, 2020-09-01:
@abdulsiddiqi alternatively, instead of relying on `bufferDurationOverflowPressure` callback which is called when it reaches 95% percentage (static). You could try to use the latency pressure callback instead which is configurable depending on your max latency.

MushMal, 2020-09-02:
@abdulsiddiqi there are many issues that I would like to address one-by-one 

* The content store is natively allocated. By default, it's implemented in RAM using low-fragmentation fast heap. No JVM.
* Your configuration is VERY odd. You specify 17 seconds of temporal buffer with 14 seconds of latency pressure. This is too aggressive with specified 2MBps average streaming as any hiccup in the network stack will cause a cascading effect - for example, when the session expires and we need to do switching to a new session. Recommend increasing the temporal buffer. 
* You have specified a 1 second timecode scale. This means that the generated MKV will have a second granularity. This parameter is intended to be used by really advanced users who not only understand the deep intricacies of the MKV encoding but also the implications on the underlying elementary stream. For vast majority of the audio/video streaming cases the default 1ms timecode scale is perfect.
* You have also specified the network staleness to 9seconds. This might be too aggressive when going over the public internet, though, not out of norm. Recommend specifying something that will be specific to the jitters of your network. In general public internet might tolerate 20 seconds of delay in the acks.
* Recommend thinking through your application and look into what your application needs are. If you really intend to use java to stream true real-time stream and have 14 seconds of latency requirement? Java GC itself might take way over that. Are you sure you are looking at the right solution in that case? I have no context as to what your application is attempting to do but suggest looking at WebRTC (also available from KVS) and other realtime streaming solutions.

abdulsiddiqi, 2020-09-03:
There was a typo in the timecodeScale value, its the default value of 1ms. 

* What action items are taken from the KVS client when the latency pressure is fired? Will just the callback just be fired?
* Lets say I set the connection staleness to 20 seconds. That should decrease the likelihood of the connection staleness callback getting fired. What other behavior would change in the client?

MushMal, 2020-09-03:
Java SDK will not attempt to do anything on these callbacks. It’s up to the application to handle them. Some applications simply ignore those. In case of the buffer pressure and latency pressure, this can lead to frames being dropped eventually when the buffer fills up. Stream stale callback is slightly different, it tells that the upload of the frames is ongoing but it hasn’t received an ACK for the duration that’s specified in the streaminfo structure. 

Applications have a wide variety of choices on how to handle these situations. Some can terminate and free the stream and retry. Others can also dynamically reduce the frame rate for example to help to handle the temporary network pressures

disa6302, 2020-09-21:
Closing assuming question is answered.

closed 2020-09-21 by disa6302