awslabs/amazon-kinesis-video-streams-producer-sdk-java #121: Not abel to exteact cpd data from camera frames in java producer SDK

amiya-elear, 2020-09-09:
Hi Team,

I am generation frames from the camera and directly stream those frames to kinesis while I am using C producer SDK its able to extract the CPD data from I frames and it's streaming perfectly no requirement of providing CPD externally, but when I am using java producer SDK MultiTrackFrameSource to stream data in kinesis it's asking me to send CPD in track info.
I believe that there must be some flag that I am missing in my java code to extract CPD from the frame. Can you guys help to resolve the issue?

Here are my camera frames
https://drive.google.com/file/d/1b6COse4HYVjJG07fPTdlyofMgoSGby-z/view?usp=sharing 

Thanks

MushMal, 2020-09-09:
The code is the same. Please make sure you update to the latest SDK. Ensure you use the right adaptation flags in the stream info - those should state to adapt from Annex-B to AvCC. The H264 (H265) frames have to be in Annex-B format, the I-frame should have CPD prepended - in case of H264, it should be SPS/PPS combination.

amiya-elear, 2020-09-09:
Yes, In my MediaSource I use proper adaptation flags now its ability to extract

closed 2020-09-09 by amiya-elear