awslabs/amazon-kinesis-video-streams-producer-sdk-java #123: Failed to put a frame with status code 0x52000052

amiya-elear, 2020-09-10:
Hi Team,

My scenario:
1: I am creating a stream and keeping it ready to stream the data into kinesis.
2: any request from the user will start the media stream MediaSource.start() wait for a callback, whenever I will get an onDataReceive callback, Then the kinesis listener will put frame into the kinesis stream.
3: I am stoping the stream after 5 mints using MediaSource.start to start the stream and getting the call

createStream()->MediaSource.start()->putFrames()...->MediaSource.stop()

Issue: For the first request, flow work like a charm, for the second request, calling Mediasource.start() and getting onDataReceive callback but whenever it's trying to put the frame to kinesis getting an error: Failed to put a frame with status code 0x52000052 

MediaSource.start()->putFrames()..error:0x52000052 ...->MediaSource.stop()

I want to trigger a stream into a state that can receive the frame again after stopping it.

I did some debugging and found that after stopping the stream MediaSource sate became stopped but when I am starting the MediaSource again it's not changing the MediaSource state to ready.

Any solutions for this issue?

Thanks.

hassanctech, 2020-09-10:
I just tried this with our DemoApp and I see the same thing, I will investigate and get back to you as soon as I have something.

MushMal, 2020-09-10:
What you seem to try to achieve is an "intermittent" streaming scenario. In this case, you don't have to call stop() as after calling stop the stream can not be used to produce frames to. Stop API is used to indicate a graceful stop - it attempts to stream out the remainder of the buffer and wait for the last ACK.

In your case, you just create the stream, start, put frames until you have them then do nothing - just pause. When you have more frames, continue putting.

If you call stop in order to ensure you get the stream bits persisted before the next "session" you have to either re-create the stream or reset the stream to continue.

amiya-elear, 2020-09-10:
Thanks, Mushegh.

Can you please me with how to reset the stream in kinesis java producer SDK?

MushMal, 2020-09-10:
Hmm, it seems that the reset stream API has not been exposed to Java layer...

amiya-elear, 2020-09-10:
Yes, I have just checked its not exposed.

MushMal, 2020-09-10:
Added a link to a work item: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/issues/124

MushMal, 2020-09-10:
For now, you could simply do create/putframe.../stopsync/free repeat when the new frames are available

amiya-elear, 2020-09-11:
Ok Thanks, Mushegh, So I have to create the stream every time after stoping it.

MushMal, 2020-09-11:
@amiya-elear when you have an intermittent producer, you could simply do nothing. The flow will look something like this:

Create/putFrame..../pause for a while/putFrame..../StopSync/Free 

Basically, just pause when you have no more frames and not stop it. In this case, what would happen is that the underlying connection will timeout if no data is exchanged within a 30 seconds. When you put frames for the new session, the state machine will issue a new PutMedia request and will start streaming. 

Important fact here. If you have configured your stream with persistence (retention period != 0) then when the connection gets terminated on timeout, the low level wouldn't know whether the fragment data has been successfully persisted. This is due to the fact that the backend doesn't have any way of knowing that the stream has been paused - or the previous fragment is finished until it gets a new fragment. This is the reason that the producer would re-send the previous fragment with the new session.

Now, there are two ways to handle this if you really need to handle the case of re-streamed last fragment of the intermittent session.

1) when you are done with the session, you could send a special EoFR frame. Take a look at the native code EOFR_FRAME_INITIALIZER and its usage in the tests: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/mkvgen/include/com/amazonaws/kinesis/video/mkvgen/Include.h#L316

2) Create separate sessions by either create/putframe/.../stopSync/Free pattern or create/putframe/.../reset stream/ continue

disa6302, 2020-09-21:
@amiya-elear , 

Closing this assuming your question is answered. Feel free to reach out if you see any issues in the future.

closed 2020-09-21 by disa6302