awslabs/amazon-kinesis-video-streams-producer-sdk-java #130: putFrame returns error with status code 0xd

abdulsiddiqi, 2020-09-23:
Error logs:

> KinesisVideo:Kinesis Video service client returned an error java.lang.RuntimeException: Exception while reading status line ! . Reporting to Kinesis Video PIC
> com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream. StatusCode: 0xd

This happened for one particular stream, while the other streams on that host were behaving just fine. The stream latency pressure callback was fired but the connection couldn't be reset. At this moment we believe restarting the whole stream is a possible fix to overcome this error. We will do this by calling unregisterMediaSource on this stream followed by registerMediaSource.

Is this is right action path when such an error occurs? Why is this error occurring in the first place?

MushMal, 2020-09-23:
Sounds reasonable, however, it would be nice to get the details on what's going on.

Could you please try to include the stream ARN/name, the time range (so we can look up in the backend), the debug logs when this happens and any general info that makes this particular stream any different - for example - could this collide with any other device upload that ends up with the same stream or something in that nature.

hassanctech, 2020-10-30:
Closing due to inactivity.  Please open a new issue if there is more help we can provide.

closed 2020-10-30 by hassanctech