awslabs/amazon-kinesis-video-streams-producer-sdk-java #131: how to get  list resolution video through DASH url from KVS stream

amiya-elear, 2020-09-30:
Hi Team,
From my camera I am streaming 1080P video in KVS. Is it possible to downscale the stream in KVS itself to provide a 720p version of the stream? Means while I am playing back the video using DASH url the resolution should downscale 720p.

Thanks,

MushMal, 2020-09-30:
@amiya-elear there is no way to simply "downscale" an elementary stream. It actually involves decoding, downscaling and then re-encoding the stream. KVS doesn't do this - moreover, we don't have access to even do it as the stream is encrypted.

The question is why do you even want to stream a dense 1080p to only downsample it later? Not knowing your actual application and requirements here are some suggestions:

* Try to configure your camera to produce 720p stream in the first place before the encoder
* If you can't configure the source (say it's a rtsp stream) then if your producer device allows then you could transcode it on the producer to avoid streaming extra bits
* Write an application that will run on the cloud and will consume and transcode the stream into a derivative stream - the one that could be used for the playback
* If you have access to the player (HLS or DASH) code and the bandwidth is not an issue then you could pull down 1080p, decode and then downsample the decoded images.

Please provide more info or resolve this issue

amiya-elear, 2020-09-30:
Thanks, Mushegh,

Now my camera is configured to stream only 1080p video to KVS. I want to stream the same video in Google nest hub unfortunately that supports only H.264 High Profile up to level 4.1 (720p/60fps) so I am unable to play the video in google Nest Hub. If KVS provides any feature to downscale the video then I can do the need full and provide 720p stream video URL to google Nest Hub to play it back.

MushMal, 2020-09-30:
Understood. Unfortunately, there is no out-of-the-box feature ready for the transcoding - as I mentioned, this is not a simple task in itself due to decoding, downsampling and encoding back but it's also due to the encrypted data. I have provided some answers above that you could follow. 

I am going to close this ticket as there are no direct actionable items that we can help you with

closed 2020-09-30 by MushMal