awslabs/amazon-kinesis-video-streams-producer-sdk-java #133: Question: Use Kinesis Video Stream lib in AWS Lambda

Julchenwsf, 2020-11-03:
Hello I have downloaded the repository. If I execute it locally everything works fine, but when I add an lambda function handler and try to use the kinesis stream library it shows the following message: 

_==================== FUNCTION OUTPUT ====================
"Failed loading native library StatusCode: 0xd"
==================== FUNCTION LOG OUTPUT ====================
START RequestId: 4de05ce2-1418-4c67-a41e-7ebaa8121acb Version: $LATEST
Input: {Records=[{eventVersion=2.0, eventSource=aws:s3, awsRegion=us-east-1, eventTime=1970-01-01T00:00:00.000Z, eventName=ObjectCreated:Put, userIdentity={principalId=EXAMPLE}, requestParameters={sourceIPAddress=127.0.0.1}, responseElements={x-amz-request-id=C3D13FE58DE4C810, x-amz-id-2=FMyUVURIY8/IgAtTv8xRjskZQpcIZ9KG4V5Wp6S7S/JRWeUWerMUE5JgHvANOjpD}, s3={s3SchemaVersion=1.0, configurationId=testConfigRule, bucket={name=sourcebucket, ownerIdentity={principalId=EXAMPLE}, arn=arn:aws:s3:::mybucket}, object={key=HappyFace.jpg, size=1024, eTag=d41d8cd98f00b204e9800998ecf8427e}}}]}WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
WARN / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
ERROR / KinesisVideo: 2020-11-03T19:19Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path: no KinesisVideoProducerJNI in java.library.path
END RequestId: 4de05ce2-1418-4c67-a41e-7ebaa8121acb
REPORT RequestId: 4de05ce2-1418-4c67-a41e-7ebaa8121acb	Duration: 3.87 ms	Billed Duration: 100 ms	Memory Size: 512 MB	Max Memory Used: 80 MB_	

**What do I have to do to get it works on AWS Lambda, instead of local ?**

Best regards Julia

MushMal, 2020-11-03:
There shouldn't be any real problems with lambda aside from the fact that it's execution is limited in time.

Check your verbose logs - from the small snippet of log it seems that you are missing the actual native library. Follow the readme steps to install/build the native components.

Julchenwsf, 2020-11-03:
I followed the readme of this repository. It worked for java locally there it finds the native library but when uploading the file it is not finding the library. Do I have to put the native library on s3?

Maybe you have a hint where I have to put the native library to get it work.

MushMal, 2020-11-03:
Per readme:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java#prerequisites

There are some pre-built binaries with the repository for different platforms:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/tree/master/src/main/resources/lib

If your platform is not the one in the list then you need to build the binary from scratch following the readme.

Julchenwsf, 2020-11-04:
I also built it under amazon ami successfully. 
This is my main:

public class JavaSdk implements RequestHandler<Object, String> {
	
	 static String result = "";
	 
	static {	
		System.load("/opt/lib/amazonami/libKinesisVideoProducerJNI.so");
		result = "Native code library loaded.\n" ;
	}
	
    @Override
    public String handleRequest(Object input, Context context) {
     try {
         KinesisVideoClient kinesisVideoClient = KinesisVideoJavaClientFactory
                .createKinesisVideoClient(
                        Regions.US_EAST_1,
                        AuthHelper.getSystemPropertiesCredentialsProvider());
        result = result + " successfull";
        }
        catch (Exception ex)
        {
        	result =  result + " error " + ex.getMessage();
        }
        return result;
    }
}

And this is the result by AWS Lambda:
ERROR / KinesisVideo: 2020-11-04T14:33Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path: no KinesisVideoProducerJNI in java.library.path

The system.load command is working. But when I try to execute the try inside "handleRequest" it does not work. 
Maybe you have a working demo with AWS Lambda for me?

Best regards 

Julia

unicornss, 2020-11-04:
Hey @Julchenwsf ,

Here are few suggestions that you could try.

1. Could you add the following to see what paths are being checked for  loading the shared library.

> `System.setProperty("jna.debug_load", "true");`

2. Alternatively, you can try setting the JVM args in command line as below:

> `java -Djava.library.path=".:/opt/lib/amazonami" JavaSdk`
> and
> `System.loadLibrary("KinesisVideoProducerJNI"); `

3. You can set the LD_LIBRARY_PATH environment variable if possible:

> ` export LD_LIBRARY_PATH=/opt/lib/amazonami:$LD_LIBRARY_PATH`

Julchenwsf, 2020-11-04:
Thanks. I checked the Java Library path and change the lib to this folder. Now the library can be found. But the next problem is that the AWS Lambda reported: `Can't load library: /opt/lib/libKinesisVideoProducerJNI.so: `

Do you have an idea why this could be?

Best regards

Julia

MushMal, 2020-11-04:
Is the library present in this path? Is this library referenced by LD_LIBRARY_PATH? Is this library built with platform specific toolchains? Can you access the symbols?
```

nm -gD libKinesisVideoProducerJNI.so
objdump -TC libKinesisVideoProducerJNI.so
readelf -Ws libKinesisVideoProducerJNI.so

```
You might want to do some debugging as it's impossible for us to tell what's going on.

Julchenwsf, 2020-11-04:
Hello,

the library is present in the path. I referenced it by LD_LIBRARY_PATH. I build it with an EC2 instance of "amazon/amzn-ami-hvm-2018.03.0.20181129-x86_64-gp2". The build process was succesful. The ldd output was:

ldd libKinesisVideoProducerJNI.so
        linux-vdso.so.1 =>  (0x00007ffe045ea000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007efefef1c000)
        librt.so.1 => /lib64/librt.so.1 (0x00007efefed14000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007efefe98f000)
        libm.so.6 => /lib64/libm.so.6 (0x00007efefe68d000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007efefe477000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007efefe25b000)
        libc.so.6 => /lib64/libc.so.6 (0x00007efefde8d000)
        /lib64/ld-linux-x86-64.so.2 (0x00007efeff37c000)

Attached there are the output of the three commands. Maybe you can see what is wrong. I don´t have deep knowledge in C++. So I don´t know what the output has to be. 

Best regards 

Julia

[out 1.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/5490173/out.1.txt)
[out2.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/5490174/out2.txt)
[out3.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/files/5490175/out3.txt)

MushMal, 2020-11-04:
Your binary looks good. Try to debug why your path is not pointing to it. Make sure you specify the correct path. There is no other thing that comes to my mind, if, your problem is related with inability to load the binary.

Try to capture the verbose level logs - perhaps you have other issues and we are chasing the wrong issue.

Julchenwsf, 2020-11-04:
How can I capture the verbose level logs?

MushMal, 2020-11-04:
Java side is using log4j. Check out the following I googled: https://www.tutorialspoint.com/log4j/log4j_logging_levels.htm

I want to narrow down and ensure the issues you are seeing are related to the library loading and not something else.

MushMal, 2020-11-08:
Any update on this issue? Seems that you need to get a little more debugging done on your platform to understand why the library is not found/loaded.

Julchenwsf, 2020-11-12:
So I tried some debugging. My approach was to run it locally on the ec2 Instance to get a more valuable error message. Now I get these error: This app is built to run with version 1.11 of the libKinesisVideoProducerJNI.so library, but version 2.0 was found on this device. So I tried to build the library with version 1.11 but unfortunately there is no version 1.11. Is there a possibilities to get version 1.11 for this library? Or can you update these project to version 2.0. 

Looking forward to your answer.

Best regards 

Julia

MushMal, 2020-11-12:
I see. In order to avoid versioning issues across the JNI layer, we version Java and the JNI counterparts separately and check the versions at runtime. So, the Java layer after loading the library calls an API to get the compile info which includes the version. If the version of the native library is older then it will fail.

You need to ensure you have the latest Java codebase and ensure you are building the latest native library from the repository. It seems that either using Mavin or something like that brought an older version of the native library.

Try to start from scratch and get it working on the stock components and only after that try to modify.

Julchenwsf, 2020-11-17:
After starting from scratch, now everything works. Thanks for your help :-)

closed 2020-11-17 by Julchenwsf