awslabs/amazon-kinesis-video-streams-producer-sdk-java #136: Push live stream to Kinesis video stream from webcam

BodapatlaKiranReddy, 2021-02-10:
How can use i use web cam to integrate with kinesis video stream producer to push the live stream?
The SDK show sample demo but how to connect with webcam  and in this SDK we don't find the CameraMediaSource class.

Any suggestion on this will really help me a lot.
Thanks in advance

BodapatlaKiranReddy, 2021-02-11:
> How can use i use web cam to integrate with kinesis video stream producer to push the live stream?
> The SDK show sample demo but how to connect with webcam and in this SDK we don't find the CameraMediaSource class.
> 
> Any suggestion on this will really help me a lot.
> Thanks in advance

@zhiyua-git 
@MushMal 

Any suggestion will be appreciated

MushMal, 2021-02-11:
CameraMediaSource is a simple helper and doesn't actually provide access to your laptops generic webcam. It's used with the Android solution where there is a standardized media pipeline (MediaCodec API). Generic Java doesn't have a common access to a/v hardware nor the encoders on the device.

Try to read through some of the posts. I just searched the keyword in the issues and pasted here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/search?q=webcam&type=issues

BodapatlaKiranReddy, 2021-02-12:
@MushMal 
Thanks for your response.
Is seems like we don't have option to stream from webcam? or do we have in java?
Any idea.

> CameraMediaSource is a simple helper and doesn't actually provide access to your laptops generic webcam. It's used with the Android solution where there is a standardized media pipeline (MediaCodec [...]

@MushMal 
Thanks for your response.
Is seems like we don't have option to stream from webcam? or do we have any other option in java?
Any idea.

MushMal, 2021-02-12:
Your question is not properly formulated. There is no Java standard common way to handle it but there are some 3rd party libraries you could use. I attached the search results so you can read through and get more info

BodapatlaKiranReddy, 2021-02-12:
> Your question is not properly formulated. There is no Java standard common way to handle it but there are some 3rd party libraries you could use. I attached the search results so you can read through and get more info

ok sure, Thanks

hassanctech, 2021-02-16:
@BodapatlaKiranReddy It looks like this question has been answered.  I am closing this issue.  If you have any more questions please open a new issue.

closed 2021-02-16 by hassanctech