awslabs/amazon-kinesis-video-streams-producer-sdk-java #153: No media found Verify that you are streaming to us-east-1 and the fragment timecodes are correct, or download the Kinesis Video Streams Producer SDK to set up your device.

ZafarQuaere, 2022-02-11:
[image]
I am getting this error on AWS console when updating from  **aws_version = '2.28.+'** to **aws_version = '2.30.+'**.
below 2.30 version I am getting an orientation preview issue [orientation issue](https://github.com/awslabs/aws-sdk-android-samples/pull/345).

niyatim23, 2023-02-06:
Hi @ZafarQuaere, apologies for the delay in response. Are you still facing this issue?

ZafarQuaere, 2023-02-07:
@niyatim23, Perhaps issue is resolved as i have moved out of this task.
However, It will be wonderful if you can provide any solution or step to resolve this issue.

niyatim23, 2023-02-07:
@ZafarQuaere, can you try using the sample application [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/src/main/demo/com/amazonaws/kinesisvideo/demoapp/DemoAppMain.java) as is and see if the issue persists?

ZafarQuaere, 2023-02-07:
Thanks @niyatim23, I will check this.

niyatim23, 2023-02-13:
Closing assuming answered. Feel free to open another issue if the problem persists