awslabs/amazon-kinesis-video-streams-producer-sdk-java #159: Push H265 stream to KVS and get it, but it fails

fankaibo, 2022-03-14:
I saw from our official documentation that KVS Video Stream is a streaming solution that supports H265+AAC. And I let my camera device call the class library of C language for track1: H265, Track2: stream information in AAC format. 
When I use SDK GetDASHStreamingSessionURL, SDK returns me: 
UnsupportedStreamMediaTypeException
The type of the media (for example, h.264 or h.265 video or ACC or G.711 audio) could not be determined from the codec IDs of the tracks in the first fragment for a playback session. The codec ID for track 1 should be V_MPEG/ISO/AVC and, optionally, the codec ID for track 2 should be A_AAC.
I have attached a screenshot of the device calling the SDK.
[image]

Please let me know where I might be going wrong, thanks a lot.

fankaibo, 2022-03-16:
Allow me to correct the description above.
We are using the combination of track1: H265, track2: G711 to push the stream, and the error described above occurs when getting the stream.
(Track2: AAC I said above, the description is wrong.)

niyatim23, 2023-02-03:
Hi, it seems like you have opened this issue in the incorrect repository. If this still persists, please open it in [this](https://github.com/awslabs/amazon-kinesis-video-streams-producer-c) repository

closed 2023-02-03 by niyatim23