awslabs/amazon-kinesis-video-streams-producer-sdk-java #172: Can't find dependent libraries

sivakrishna0807, 2022-07-29:
I am running a kinesis video streams producer sdk program on Windows10 using Eclipse IDE. But I am getting an error saying that ERROR com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory - Unsatisfied link error. java.lang.UnsatisfiedLinkError: C:\Program Files\Java\jdk-11.0.14\bin\KinesisVideoProducerJNI.dll: 
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:61)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd

I have added .dll path in kinesis native library and I have no clue of what to do next.

disa6302, 2022-08-01:
@sivakrishna0807 , 

Which library are you using? Are you using a relative / absolute path?

Are you building the JNI library on your own or using the library available in this repo here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/tree/master/src/main/resources/lib/windows

sivakrishna0807, 2022-08-01:
@disa6302 I tried using both relative /absolute path but  I am getting the same error..........and I didn't use my own own JNI library.I have simply used the JNI library given in this repo.

niyatim23, 2022-10-14:
Hi @sivakrishna0807, please make sure to NOT include the library name in the file path. Refer [this](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java/blob/master/.github/workflows/ci.yml#L45) to set the arguments properly. We have this library working in our CI so it seems like a problem with the path specified to run the application

Closing assuming answered. Feel free to reach out to us with further questions / issues.

closed 2022-10-14 by niyatim23