awslabs/amazon-kinesis-video-streams-producer-sdk-java #173: Insert audio into kinesis video

AWilson10, 2022-09-05:
Hello,
I am using the amazon-kinesis-video-streams-producer-sdk-java. 
My aim is to create a lambda that is in the amazon connect flow with audio to and from customer sent to the lambda.
My aim is to extract and insert audio in the live feed.
I have managed to extract  audio using the GetMediaWorker) and decoding for the correct audio channel.
But I want to take a separate audio feed and send it to the customer. I presumed that this would use the PutMediaWorker
but I cant see how to insert the data into the fragments correctly. 
Decoding was more straight forward. Do you have any advice on this?

irwinb, 2022-09-28:
Hi @AWilson10 ,

 For each incoming fragment, you will have to construct a new fragment containing the video, along with another track in the cluster with your audio. The producer library here doesn't have any facilities to do this out of the box.

AWilson10, 2022-09-28:
Thanks for the comment. What I am trying to achieve is to maintain the audio call in the lambda (launched from connect flow) so not sure if it's possible what I am trying to do. So a 2 way call can be maintained between a bot and a client but not using Lex, using custom audio bytes. I have not tried gstreamer as I want to maintain the customer to bot audio adn I expect gstreamer will overwrite the stream. As I say, it may not be possible.