awslabs/amazon-kinesis-video-streams-producer-sdk-java #179: errorCode=INVALID_MKV_DATA

andreikukenov, 2023-02-24:
Could you help me to figure out why my mkv video file is wrong.
mediainfo output looks like:

`General
Unique ID                                : 123230831018932709636111506672951743609 (0x5CB5650E8A2B66FCC588EA8810EFD479)
Complete name                            : /video-capture/records/low_resolution/cam1/record_1677231463791_cam1_l.mp4
Format                                   : Matroska
Format version                           : Version 2
File size                                : 1.73 MiB
Duration                                 : 30 s 0 ms
Overall bit rate                         : 483 kb/s
Movie name                               : {'start_time': 1677231463.7916667, 'interpipe': 'cam1'}
Encoded date                             : UTC 2023-02-24 09:37:44
Writing application                      : GStreamer Matroska muxer
Writing library                          : GStreamer matroskamux version 1.20.3
FileExtension_Invalid                    : mkv mk3d mka mks

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High 4:4:4 Predictive@L3.1
Format settings                          : CABAC / 3 Ref Frames
Format settings, CABAC                   : Yes
Format settings, Reference frames        : 3 frames
Format settings, GOP                     : M=1, N=12
Codec ID                                 : V_MPEG4/ISO/AVC
Duration                                 : 30 s 0 ms
Bit rate                                 : 474 kb/s
Nominal bit rate                         : 500 kb/s
Width                                    : 800 pixels
Height                                   : 600 pixels
Display aspect ratio                     : 4:3
Frame rate mode                          : Constant
Frame rate                               : 10.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:4:4
Bit depth                                : 10 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.099
Stream size                              : 1.69 MiB (98%)
Title                                    : Video
Writing library                          : x264 core 163 r3060 5db6aa6
Encoding settings                        : cabac=1 / ref=3 / deblock=1:0:0 / analyse=0x3:0x113 / me=hex / subme=7 / psy=1 / psy_rd=1.00:0.00 / mixed_ref=1 / me_range=16 / chroma_me=1 / trellis=1 / 8x8dct=1 / cqm=0 / deadzone=21,11 / fast_pskip=1 / chroma_qp_offset=4 / threads=12 / lookahead_threads=2 / sliced_threads=0 / nr=0 / decimate=1 / interlaced=0 / bluray_compat=0 / constrained_intra=0 / bframes=0 / weightp=2 / keyint=45 / keyint_min=4 / scenecut=40 / intra_refresh=0 / rc_lookahead=40 / rc=cbr / mbtree=1 / bitrate=500 / ratetol=1.0 / qcomp=0.60 / qpmin=0 / qpmax=81 / qpstep=4 / vbv_maxrate=500 / vbv_bufsize=300 / nal_hrd=none / filler=0 / ip_ratio=1.40 / aq=1:1.00
Language                                 : English
Default                                  : Yes
Forced                                   : No
Color range                              : Limited
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709`

I get the following error when I try to put video file to kinesis 
onAckEvent AckEvent{ackEventType=ERROR, fragmentTimecode=null, fragmentNumber='null', errorCode=INVALID_MKV_DATA, errorId=4006}

hassanctech, 2023-03-06:
Can you post the full logs?  If you use a tool like `mkvinfo` to inspect it, does it report any errors?

andreikukenov, 2023-03-13:
Hello @hassanctech please find logs below

`+ EBML head
|+ Document type: matroska
|+ Document type version: 2
|+ Document type read version: 2
+ Segment: size 15546793
|+ Seek head (subentries will be skipped)
|+ EBML void: size 142
|+ Tracks
| + Track
|  + Track number: 1 (track ID for mkvmerge & mkvextract: 0)
|  + Track type: video
|  + Track UID: 11633768198446771190
|  + Default duration: 00:00:00.033333333 (30.000 frames/fields per second for a video track)
|  + Name: Video
|  + Video track
|   + Pixel width: 1920
|   + Pixel height: 1200
|   + Display width: 1920
|   + Display height: 1440
|   + Interlaced: 2
|   + Video color information
|    + Color range: 1
|    + Color matrix coefficients: 1
|    + Color transfer: 1
|    + Color primaries: 1
|  + Codec ID: V_MPEG4/ISO/AVC
|  + Codec's private data: size 43 (H.264 profile: Baseline @L5.0)
|+ Cluster`

`[DEBUG] [13-03-2023 15:36:43:498.356 GMT] describeStreamCurlHandler(): DescribeStream API response: {"Message":"The requested stream is not found or not active."}
[INFO ] [13-03-2023 15:36:43:499.144 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [13-03-2023 15:36:43:499.390 GMT] defaultStreamStateTransitionHook(): Stream state machine retry count: 0
[DEBUG] [13-03-2023 15:36:43:499.446 GMT] defaultStreamStateTransitionHook(): 
 KinesisVideoStream base result is [404]. Executing KVS retry handler of retry strategy type [1]
[DEBUG] [13-03-2023 15:36:43:500.139 GMT] createStreamCurlHandler(): createStreamCurlHandler response 0x7f32980a86e0
[INFO ] [13-03-2023 15:36:44:413.944 GMT] writeHeaderCallback(): RequestId: 81621ce9-3a78-4ae4-965c-bdb7e3ba451b
[DEBUG] [13-03-2023 15:36:44:414.119 GMT] createStreamCurlHandler(): CreateStream API response: {"StreamARN":"arn:aws:kinesisvideo:us-west-2:676440762571:stream/test_stream/1678721804436"}
[INFO ] [13-03-2023 15:36:44:414.960 GMT] createStreamResultEvent(): Create stream result event.
[INFO ] [13-03-2023 15:36:45:168.251 GMT] writeHeaderCallback(): RequestId: 31a39ee9-0748-495b-b489-4485730f403b
[DEBUG] [13-03-2023 15:36:45:168.427 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-b1731d75.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [13-03-2023 15:36:45:169.084 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [13-03-2023 15:36:45:169.203 GMT] getStreamingTokenHandler invoked
[DEBUG] [13-03-2023 15:36:45:169.238 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1678721805169230337 Expiration: 18446744073709551615
[INFO ] [13-03-2023 15:36:45:169.272 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [13-03-2023 15:36:45:169.338 GMT] streamReadyHandler invoked
Stream is ready
[INFO ] [13-03-2023 15:36:45:171.530 GMT] kinesisVideoStreamFormatChanged(): Stream format changed.
[DEBUG] [13-03-2023 15:36:45:172.189 GMT] Dropping frame with flag: 1600
[INFO ] [13-03-2023 15:36:45:172.558 GMT] putStreamResultEvent(): Put stream result event. New upload handle 0
[WARN ] [13-03-2023 15:36:45:183.042 GMT] notifyDataAvailable(): Failed to un-pause curl with error: 43. Curl object 0x7f32980a08a0
[DEBUG] [13-03-2023 15:36:45:183.719 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:184.177 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:184.528 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:184.828 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:185.370 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:185.931 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:186.393 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:186.922 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:187.760 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:188.284 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:188.764 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:189.650 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:190.124 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:190.861 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:191.549 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:191.978 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:192.587 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:193.105 GMT] Dropping frame with flag: 1536
[... 99 lines cut ...]
[DEBUG] [13-03-2023 15:36:45:286.030 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:286.870 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:289.372 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:291.356 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:294.397 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:296.621 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:298.714 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:299.179 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:300.051 GMT] Dropping frame with flag: 1536
[DEBUG] [13-03-2023 15:36:45:302.108 GMT] Dropping frame with flag: 1536
[INFO ] [13-03-2023 15:36:45:305.553 GMT] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 0000563886773ba0.
[... cut ...]
[INFO ] [13-03-2023 15:36:49:308.012 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [13-03-2023 15:36:49:308.123 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [13-03-2023 15:36:49:709.411 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [13-03-2023 15:36:49:709.625 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
[INFO] kvs_gstreamer_file_uploader_sample: Persisted successfully. File: /home/adndrei/record_1677492535171_cam_left_src_l.mkv`

sirknightj, 2026-02-02:
It looks like a timecode issue.

Dropping frame with flag: 1536

Sometimes, GStreamer doesn't pipe the timecode to the kvssink, you can try switching the mode to offline mode.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/discussions/1008

closed 2026-02-02 by sirknightj